martinville/solarsynkv2 #1: Feature Request: Inverter settings

slipx06, 2023-11-21:
Just checking if you plan on pulling inverter settings. This will give you access to additional sensors

`curl -s -X GET -H "Content-Type: application/json" -H "authorization: Bearer $ServerAPIBearerToken" https://api.sunsynk.net/api/v1/common/setting/$sunsynk_serial/read  -o "settings.json"`

There are 6 each of these i.e. sellTime1, SellTime2 ...sellTime6, time1on-time6on, cap1-cap6

```
prog1_time=$(jq -r '.data.sellTime1' settings.json); 
prog1_charge=$(jq -r '.data.time1on' settings.json); 
prog1_capacity=$(jq -r '.data.cap1' settings.json); 
battery_shutdown_cap=$(jq -r '.data.batteryShutdownCap' settings.json); 
use_timer=$(jq -r '.data.peakAndVallery' settings.json); 
priority_load=$(jq -r '.data.energyMode' settings.json); 
```

martinville, 2023-11-22:
Don't see why not!
Will take a closer look at these to understand what they do and then add them to the entities.

martinville, 2024-01-04:
FYI I added the following data points. 

prog1_time=$(jq -r '.data.sellTime1' settings.json); 
prog2_time=$(jq -r '.data.sellTime2' settings.json); 
prog3_time=$(jq -r '.data.sellTime3' settings.json); 
prog4_time=$(jq -r '.data.sellTime4' settings.json); 
prog5_time=$(jq -r '.data.sellTime5' settings.json); 
prog6_time=$(jq -r '.data.sellTime6' settings.json); 

prog1_charge=$(jq -r '.data.time1on' settings.json); 
prog2_charge=$(jq -r '.data.time2on' settings.json);
prog3_charge=$(jq -r '.data.time3on' settings.json);
prog4_charge=$(jq -r '.data.time4on' settings.json);
prog5_charge=$(jq -r '.data.time5on' settings.json);
prog6_charge=$(jq -r '.data.time6on' settings.json);

prog1_capacity=$(jq -r '.data.cap1' settings.json); 
prog2_capacity=$(jq -r '.data.cap2' settings.json); 
prog3_capacity=$(jq -r '.data.cap3' settings.json); 
prog4_capacity=$(jq -r '.data.cap4' settings.json); 
prog5_capacity=$(jq -r '.data.cap5' settings.json); 
prog6_capacity=$(jq -r '.data.cap6' settings.json); 

battery_shutdown_cap=$(jq -r '.data.batteryShutdownCap' settings.json); 
use_timer=$(jq -r '.data.peakAndVallery' settings.json); 
priority_load=$(jq -r '.data.energyMode' settings.json); 

They all follow the usual entity naming convention. 
I.E 
sensor.solarsynk_serialnumber_prog1_time
sensor.solarsynk_serialnumber_prog2_time
sensor.solarsynk_serialnumber_prog3_time
sensor.solarsynk_serialnumber_prog4_time ect.....

closed 2024-02-12 by martinville