martinville/solarsynkv2 #3: using solarsynk with pv_opt

ElectricMolehill, 2024-03-08:
Hi Martin,

I like solarsynk because it doesn't require additional wiring to the inverter. I would like to use it with fboundy's PV optimiser for Home Assistant (called pv_opt in Github). He's looking at the feasibility at the moment. Before we get too far, can I ask if you think this is a good idea? One thing I think could be a problem is solarsynk's method for updating inverter settings. It definitely works as I have tried it myself and found that changes sent to the API are pushed to the inverter almost straight away. But for a Home Assistant extension to use it is tricky because there is a 5 min wait for the solarsynk process to run, and the response is recorded in the log where it is not very accessible. In order to confirm it has worked we then have to wait another 5 mins to see the sensors update. Would it be possible to provide a more real time update with immediate response?

cheers
Mark

alz41, 2024-03-09:
following developments

martinville, 2024-03-10:
Hi Mark. I will take a look at your request to see what it possible. 

It may be a little tricky because this addon does not have any listeners. It will fetch information on a set interval as opposed to waiting for a request. This is because its merely a bash script executed in a loop without any services.  The idea behind Solarsynk was for it to be as simple as possible without having to install any additional components. In my opinion there are too many add-ons that have a dependency on other components to make them work.

FYI: You don't have to wait 5 minutes you can set your sunsynk account on the sunsynk portal to refresh as low as 60 seconds.
But I understand the frustration of  having to wait  that long to get a response.

I believe its possible for add-ons to host a web service. Perhaps this can be added to this addon and then have it action commands as they come in.