martinville/solarsynkv2 #4: JQ issue

kitoban, 2024-03-13:
`jq: error: Could not open file settings.json: No such file or directory`

Within logs all values returning as null.

```
Core 2024.3.0
Supervisor 2024.03.0
Operating System 12.0
Frontend 20240306.0
```

Unsure on trigger for issue. currently blocking data reads.

kitoban, 2024-03-13:
Relevant logs in full:
```
/run.sh: line 121: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 122: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 124: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 125: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 126: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 127: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 128: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 129: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 132: [: ==: unary operator expected
jq: error: Could not open file batterydata.json: No such file or directory
/run.sh: line 133: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 134: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 135: [: ==: unary operator expected
jq: error: Could not open file loaddata.json: No such file or directory
/run.sh: line 136: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 137: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 138: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 139: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 140: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 141: [: ==: unary operator expected
jq: error: Could not open file griddata.json: No such file or directory
/run.sh: line 142: [: ==: unary operator expected
jq: error: Could not open file outputdata.json: No such file or directory
/run.sh: line 143: [: ==: unary operator expected
jq: error: Could not open file outputdata.json: No such file or directory
/run.sh: line 144: [: ==: unary operator expected
jq: error: Could not open file outputdata.json: No such file or directory
/run.sh: line 145: [: ==: unary operator expected
jq: error: Could not open file outputdata.json: No such file or directory
/run.sh: line 146: [: ==: unary operator expected
jq: error: Could not open file loaddata.json: No such file or directory
/run.sh: line 147: [: ==: unary operator expected
jq: error: Could not open file loaddata.json: No such file or directory
/run.sh: line 148: [: ==: unary operator expected
jq: error: Could not open file loaddata.json: No such file or directory
/run.sh: line 149: [: ==: unary operator expected
jq: error: Could not open file loaddata.json: No such file or directory
/run.sh: line 150: [: ==: unary operator expected
jq: error: Could not open file loaddata.json: No such file or directory
/run.sh: line 151: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 152: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 153: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 154: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 155: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 156: [: ==: unary operator expected
jq: error: Could not open file pvindata.json: No such file or directory
/run.sh: line 157: [: ==: unary operator expected
jq: error: Could not open file inverterinfo.json: No such file or directory
/run.sh: line 158: [: ==: unary operator expected
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file settings.json: No such file or directory
jq: error: Could not open file dcactemp.json: No such file or directory
jq: error: Could not open file dcactemp.json: No such file or directory
```

kitoban, 2024-03-13:
Done some diagnosis on this and I can't seem to establish a bearer token off of what used to work:
`SSL peer certificate or SSH remote key was not OK`
Copying the curl from the code and trying the same seems to lead to the same issue, would suspect Sunsynk have done an update on their token endpoint.

martinville, 2024-03-13:
Its also failing on my side.
Yes looks like there was a change in their API. Will resolve ASAP

martinville, 2024-03-13:
Problem resolved it self. The sunsynk server was down again today

kitoban, 2024-03-13:
Has resolved for me as well, guess it might be worth a check on bearer token validity to get a more sensible error back but beyond that can close this one