martinville/solarsynkv2 #6: Intermittent data

mdeakin99, 2024-03-27:
I have noticed that the data feed is intermittent over the last two days,  this means that sometimes the battery discharge sometimes say 0w as well as the SOC (picture attached).

Have I configured something incorrectly or is this an issue on the SunSynk Cloud side.

[image]

mdeakin99, 2024-03-27:
Just realised this is probably related to https://github.com/martinville/solarsynkv2/issues/5 now I have looked through the error log.  Feel free to close if you want to :)

martinville, 2024-04-01:
Yes, its related to Issue #5 , I made some changes to relieve most of the problems, please check #5 or the addons change log to see what was changed.

martinville, 2024-06-18:
No longer an issue was addressed in an update

closed 2024-06-18 by martinville