martinville/solarsynkv2 #7: Connection issues

alz41, 2024-03-30:
Wondering if this could be the cause of the connection issues?

Error Connecting: HTTPSConnectionPool(host='api.sunsynk.net', port=443): Max retries exceeded with url: /api/v1/plants?page=1&limit=10&name=&status= (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1000)')))

martinville, 2024-04-01:
It certainly is the issue. Initial thoughts were that there are many connection timeouts but its not the case. Curl randomly struggles to verify the validity of the sunsynk server's certificate which was renewed on the 13th of march.

I'm now in the process to change the script to not post or attempt to post data which it could not retrieve. this way it will prevent the addon from messing up graphs due to invalid data.

Its also possible to let curl ignore certificate errors. Not ideal because it then ignores the reason why certificates exist in the first place.

martinville, 2024-04-01:
Check #5 for the solution

ElectricMolehill, 2024-04-01:
Could it be that the new sunsynk.net SSL certificate is using a CA or intermediate certificate which the Solarsynk add-on doesn't have in its store and therefore can't verify? (That's the limit of my knowledge by the way - I've no idea where a docker keeps its CA certificates or how they get there in the first place!) Since the error sounds fatal it's a bit surprising that it works at all but perhaps sunsynk.net has a load balancer which has some nodes still on a valid but old certificate?
By the way, new version of Solarsynk is looking good for continuity of data but I'll let it run for a bit before confirming

ElectricMolehill, 2024-04-01:
Certificate installation at sunsynk.net looks good - all green - but perhaps we don't always get the same cert
[image]
Has Solarsynk got access to the 2 certs highlighted?

martinville, 2024-06-18:
Cert issue was fixed also , sunsynk them selves configured their certs correctly.