martinville/solarsynkv2 #8: No entities in Home Assistant

chris2172, 2024-04-15:
I am running version 2.1.14 for the first time, enabled the Long-lived Access token and added it with the Home assistant details.
In the logs all I see is 1 attempt at updating the entities.

prog5_capacity: 15
prog6_capacity: 15
battery_shutdown_cap: 15
use_timer: 1
priority_load: 1
dc_temp: 25.3
ac_temp: 26.6
------------------------------------------------------------------------------
Attempting to update the following sensor entities
Sending to http://191.168.1.192:8123
------------------------------------------------------------------------------

It has been like this overnight.
I don't see any entities with these names being available in home assistant.

martinville, 2024-04-15:
Hi Chris.
If you see values populated like you do in the above example it means that it successfully managed to grab the data from the sunsynk server so that part looks OK.

After it completed fetching info from sunsynk it will attempt to push them to home assistant.
If its getting stuck at posting to HA  and  it was working before and it just stopped. Could you double check the long lived token?
Perhaps it expired or it was removed accidentally.  Try generating a new key and update the Solarsynk config to see if that resolves the issue.

chris2172, 2024-04-16:
This is a fresh install after someone told me about this Add-on so it has never populated entities in Home Assistant.
I was updating the required fields in YAML. For some reason, the Long Live Access token was not recognised when updating it in YAML. But adding the Long Live Access token in the GUI worked fine after that.

closed 2024-04-17 by chris2172