martinville/solarsynkv2 #10: solarsynk data

harwoodrd, 2024-05-02:
All, i have set up (obviously incorrectly) the Device, and when i go to log its getting the data, however when it tries to update the entities  it fails, the log script suggested adding a helper but that shows null, can anyone point me in the right direction please, it seems so close as i can physically see the data in the log but it is just not reaching the entities...Thanks in advance

chris2172, 2024-05-02:
I had a similar issue when I added the Long-lived access token under yaml, by updating it in the GUI it worked.

Long-lived access token
In Home-Assistant click your name at the bottom
Select Security Tab
Scroll to Long-Lived Access Tokens
Click Create Token and call it SolarSynk
Copy the code.

Go to solarsynk Add-On and put it in the configuration --> 
[image]

harwoodrd, 2024-05-02:
Thank you very much for trying to assist however that did not work either, it is so frustrating to see the data you want in the log but it wont write it to the entitity, someone sugested setting up a helper but i dont know its correct format. regards Richard
 

Richard D Harwood

 
 

Sent: Thursday, May 02, 2024 at 10:30 AM

 

2024-05-02.10.29.52.png (view on web)

martinville, 2024-05-02:
Just to clarify the message letting you know to create the helper entity is optional its not a critical error and you dont have to create it.  Its only necessary if you want to send inverter settings back to the inverter. So in short it does not  have anything to do with your problem.

If the values are showing then at least the part where its fetching from the sunsynk cloud is working.

If you have a way for me to remotely log into your system I will be happy to assist. Else perhaps add a screenshot of the configuration (With the serial removed offcoarse)

brashumza, 2024-06-27:
Good day

I'm new to home assistant I've tried everything to fix this issue but with no luck can you kindly assist: see the error below

Helper entity input_text.solarsynk_000000000_inverter_settings has no value. Therefore no inverter setting will be sent for change.

martinville, 2024-06-27:
Hi there.

Its not an error. If you set the entity with a valid jason string it will
send that value to the inverter and then clear it out again.

There after you will get the same message.

 It just tells you that there are no messages to be sent.

You can safley ignore it.

brashumza, 2024-06-27:
Hi

Thanks for the response.

My problem is I'm not getting any feedback information from the inverter.

What could be the problem I've triple check my configuration and everything looks to be in order.

Your guidance to resolving this issue would be much appreciated

martinville, 2024-11-11:
Closing for now, the message below means its setup correctly.

Helper entity input_text.solarsynk_0000000_inverter_settings has no value. Therefore no inverter setting will be sent for change