martinville/solarsynkv2 #11: 3 MPPTs

AvSingh75, 2024-05-07:
Hey Martin, Great work. Loved your video and the step by step instruction. I have got everything working except that I have a 16KW Sunsynk inverter and have 3 MPPTs. Could you please make a provision for one more.

Thank you in advance

swissmike64, 2024-08-10:
Hi Martin, me too, also have the same inverter with 3 MPPT.

Could you add support for 3 and 4 strings, please?

All the best,
Michael

rkflyer, 2024-09-20:
I believe it already has support for multiple MPPT, I have 4 configured

When editing the dashboard look under solar options, change the slider for number of MPPT's and label as required.

Then under the entities section you need values from the inverter for pv1_power_186, pv2_power_187, pv3_power_188 etc...

martinville, 2024-10-13:
Hi I know its a bit late but I added mppt 3 and 4.

closed 2024-10-13 by martinville

swissmike64, 2024-10-13:
Hi, thanks so much. i appreciate your work 🙏🙏🙏