martinville/solarsynkv2 #15: Update Inverter Settings

Tabraize, 2024-07-01:
Hi Martin,

I am trying to update inverter settings but can't see how to do it. Is there any documentation for it? I noted in updates on january 3rd you mentioned you had added the functionality, Would love to get this working and honestly if it works as expected you should update the youtube video if you get chance as its a fantastic feature that im sure many people would love. The main functionality im looking for is changing the import/export of my battery settings depending on a host of factors from HA.

NoMhawk, 2024-08-08:
Also hoping to get this working.

I have created the settings helper and have tried putting in changes but get the error in the log.

XXXXX is the serial number of the master

------------------------------------------------------------------------------
Reading settings entity -> solarsynk_XXXXX_inverter_settings
------------------------------------------------------------------------------
Updating Helper: input_text.solarsynk_XXXXX_inverter_settings with: prog1_capacity: 50
{
  "code": 500,
  "msg": "Internal Server Error",
  "data": null,
  "success": false
}
Clearing previously set temporary settings.

NoMhawk, 2024-08-09:
Nope, haven't had time to work out where the 500 error is coming from, that's normally a webpage error.

ando2040, 2024-09-20:
+1 for any instructions on how to use this helper please, it sounds intriguing and as if it might be able to avoid RS485/ESP32 mods as advised on a certain South African YouTube channel?

ando2040, 2024-09-20:
OK so there is an example JSON which I assume is for the helper text field.
savesettings_example.json

My big question is how I know what all these are set to for my system, so that I don't change something unintended? Can we send just one field?

ando2040, 2024-09-22:
So I'm doing it the trial and error way.

JSON text for helper (manually via Developer tools Set State for now)
```
{
"time1on": true,
"cap1": "80",
"sellTime1": "01:00"
}
```
Gives the message in the log:
```
Updating Helper: input_text.solarsynk_xxxxxxxxxx_inverter_settings with: {     "time1on": true,     "cap1": "80",     "sellTime1": "01:00" }
{
  "code": 0,
  "msg": "Success",
  "data": null,
  "success": true
}
```

However, only 2 of my intended settings were updated. So I must have guessed the wrong setting for the first?

[image]

ElectricMolehill, 2024-09-22:
@ando2040 I've done some testing of the System mode settings. The json you sent
{
"time1on": true,
"cap1": "80",
"sellTime1": "01:00"
}
is fine but there is a special rule that time1on (Grid charge timer 1) and genTime1on (Gen charge timer 1) must be sent together, even though you only want to change one of them. Try this:
{
"time1on": true,
"genTime1on": false
"cap1": "80",
"sellTime1": "01:00"
}
NB I think the Home Assistant helper is limited to 200 characters

Some of your other questions:
**sysWorkMode**
It's clearer on the app what the settings do but according to my testing (On inverter, Z = Zero export, S = Solar export, L = Limit to load only):
sysWorkMode = 0
Z=0, S=0, L=0
App setting: Selling first (ie force export of battery to target SOC)

sysWorkMode = 1
Z=1, S=1, L=1
App setting: Zero Export + Limit to Load only
Also sets Export Solar = Y

sysWorkMode = 2 
Z=1,S=1,L=0
App setting: Limited to home
Also sets Export Solar = Y

**solarMaxSellPower**
This is a bit confusing, in fact it might be a mistake in the API. 
solarMaxSellPower is mapped to Max Solar Power in Advanced settings
pvMaxLimit is mapped to Export Power Limiter on the inverter,  Max Sell Power in app. This can only be set in increments of 10A

**energyMode**
energyMode = Priority on inverter, Energy Pattern in app
0 = Priority Batt
1 - Priority Load
I haven't tested this fully but it is something like
0 = Use Solar for battery first, then Load
1 = Use Solar for Load first, then battery

ando2040, 2024-09-22:
Brilliant, thanks! I was planning to test genTime1on but I'm not sure I'd have sent it at the same time as time1on.

ando2040, 2024-09-23:
Ha, I didn't check your JSON properly and inadvertently created the 500 error in the log that @NoMhawk and @boereworswizard had,

```
Updating Helper: input_text.solarsynk_xxxxxxxxxx_inverter_settings with: { "time1on": true, "genTime1on": false "cap1": "80", "sellTime1": "01:00" }
{
  "code": 500,
  "msg": "Internal Server Error",
  "data": null,
  "success": false
}
```

It's simply missing a comma after false. A JSON checker (eg https://jsonlint.com) will highlight it if you can't spot it. 

```
{
    "time1on": true,
    "genTime1on": false,
    "cap1": "80",
    "sellTime1": "01:00"
}
```

Still only updates the last two fields though - more experimentation required.

ando2040, 2024-09-23:
I apologise; it doesn't matter what you set the helper length to, the value becomes unknown if you try more than 200 chars!

I'm still struggling, I sent a more complete message but the checkbox remains resolutely unchecked.

```
{
    "time1on": true,
    "genTime1on": false,
    "sellTime1": "01:00",
    "sellTime1Pac": "12000",
    "cap1": "80"
}
```

ElectricMolehill, 2024-09-23:
Just tried it again  this worked for me:
{
"time1on": true,
"genTime1on": false
}

ando2040, 2024-09-24:
Thanks. I was hopeful that less could be more, so I sent just the two. Still no change. 

Now, I do find the other options confusing, could the setting be dependent on the ones above? I don't think so because the GUI allows me to check Grid Time1 and save, but the API is not taking it. 

[image]

ElectricMolehill, 2024-09-28:
Sorry I didn't get any further. I copied and pasted your json into my helper and it worked. The only other things I can think of are Use Timer {"peakAndVallery": "1"} (which is shown on your screenshot as already set), or some invalid settings on Timer 1 - but I think this is difficult to do unless the 6 Timers span more than 1 x 24 hour period and in any case if you can set Timer 1 manually it's hard to see why the API should refuse to set it. My only other observation is that your inverter looks like a different model to mine. Mine is an ECO 3.6kW but your max power is set to 13.4kW so is presumably a bigger device. Again it's hard to see why this would make a difference since the API is the same for both and the firmware can't be much different

martinville, 2025-03-21:
I’ve moved this entire add-on to a more suitable development environment, which will enable sending settings to the Sunsynk server moving forward.

The core issue, as some have already pointed out, is that the Sunsynk API expects a JSON request containing all settings for a particular module. Unfortunately, it will reject the request if only one setting is sent. The problem here is that Home Assistant’s Entity only allows a limited number of characters, whereas the settings JSON exceeds this character limit.

The solution is to check the settings in the entity, then on the backend, rebuild a template settings JSON by downloading the current settings, populating the template, injecting the settings from the entity, and then posting them to Sunsynk.

I’ve already started working on this, but I expect there may be some minor issues after launching Sunsynk v3. In the meantime, I’ll do my best to get the settings posting functionality out as soon as possible.

See the new addon --> https://github.com/martinville/solarsynkv3

martinville, 2025-03-21:
I'm closing this issue for now, as there is nothing more I can do to fix it within this repository.

closed 2025-03-21 by martinville