martinville/solarsynkv2 #16: Additional entities

brashumza, 2024-07-03:
Hi Martin

thanks for the great inverter integration, which is very user friendly even us beginners

I just want to ask if it's possible to request additional entities from the inverter. 

If yes, can you kindly point me in the right direction on how to go about it?

imtiyaazd, 2024-07-30:
Thanks for great add-on. 

I am a newbie and have a 3phase inverter. I’d like to understand how to add the “entities” for each grid phase. The grid power you pull in is the 1st phase only. 

Any help would be appreciated. 

thanks again

GrayC88, 2024-08-02:
Thank you for the Great add-on.

Could I request that we can have a non-essential load sensor and CT sensor.

Many Thanks

GrayC88, 2024-11-11:
Im still interested in the Non Essential Power
[image]