martinville/solarsynkv2 #18: /run.sh: line 401: [: ==: unary operator expected

tarranjones, 2024-08-15:
Hi 

not sure what i did wrong but i'm getting the following error 
`/run.sh: line 401: [: ==: unary operator expected`
https://github.com/martinville/solarsynkv2/blob/6b7b7e456fe15706300a4f1a6c67e73f2f4e67c2/run.sh#L401

```
------------------------------------------------------------------------------
Attempting to update the following sensor entities
Sending to http://http://homeassistant.local:8123/:8123
------------------------------------------------------------------------------
------------------------------------------------------------------------------
Reading settings entity -> solarsynk_*****_inverter_settings
------------------------------------------------------------------------------
/run.sh: line 401: [: ==: unary operator expected
Helper entity input_text.solarsynk_*****_inverter_settings has no value. Therefore no inverter setting will be sent for change.
Clearing previously set temporary settings.
Fetch complete for inverter: *****
All Done! Waiting  300  sesonds to rinse and repeat.
```
also just noticed "sesonds" is a typo,

tarranjones, 2024-08-15:
spotted the issue 

Sending to **http://http://homeassistant.local:8123/:8123**
i had filled out the host with the full url.

Still not sure why it caused a bash error.

martinville, 2024-08-22:
There isn't any proper error checks. So sometimes it will behave illogically.

Glad you came right.
Cheers

closed 2024-08-22 by martinville