martinville/solarsynkv2 #20: Loss of connection to SunSynk server causes Energy data issues

jeybee74, 2024-10-24:
Hi,
The integration works great, but I often get wrong data in my Energy Chart, this is caused by some connection problems to the SS server I guess...?

[image]

[image]

Is there a quick fix so this, so I don't have to manually fix the Statistics for this data "all the time"
:)

martinville, 2024-11-11:
Hi I'm afraid I don't have a solution for you. Quite often the SunSynk server becomes unavailable for the logger but is still available via API, The addon is still able to gather info but 0 values are served.

At the moment Its not possible for the addon to know when a value is "really" 0.

Sorry about that,

closed 2024-11-11 by martinville

martinville, 2024-12-20:
> @martinville <https://github.com/martinville> I know you closed this. But
> could it not check the last value on an incrementing total.
> If this not 0, but now reports 0 then it must be wrong (possibly with a
> unless it's midnight or an unless I get 10 zeros in a row)

Hi Chris.

I see you requested some changes. I have not tested any of it, I hardly use
this integration i since moved over to Modbus over Rs232 using an ESP32.

I haven't tested any of the changes you requested, Can you confirm they
work ? Then I will just accept the changes.

BlythMeister, 2024-12-20:
I'm running off my fork at the moment, it seems ok.  But maybe hold off just to be sure.  I'll comment on the PRs when I'm confident?

martinville, 2024-12-20:
Ok great.

Thanks for the contribution.