martinville/solarsynkv2 #22: Inverter Data not showing correctly.

GrayC88, 2024-11-11:
Good day,

I have notice that the Inverter Power, Inverter Current and Inverter Voltage are not showing the correct data. 

[image]
[image]
[image]

martinville, 2024-11-11:
Hi 
Please check the SolarsynkV2 log. In my case there was an API error caused due to an HA Core update. After updating to the latest update a few seconds ago it started to work again.

Perhaps your values are not updating due to the initial update which broke some addons.?

GrayC88, 2024-11-11:
Hi,

I have looked at the logs and I don't see any errors. I did notice that I have the following in the logs:

inverter_frequency 50.2
inverter_power 1371
inverter_current 5.9
inverter_voltage 238.9
inverter_power1 0
inverter_voltage1 0
inverter_current1 0
inverter_power2 0
inverter_voltage2 0
inverter_current2 0

But if I look in the entities I only have Inverter Power2

It looks like they are all coming through with the same entity ID

sensor.solarsynk_X_inverter_current    3.5
sensor.solarsynk_X_inverter_power      808
sensor.solarsynk_X_inverter_voltage    239.1
sensor.solarsynk_X_inverter_current    0
sensor.solarsynk_X_inverter_power      0
sensor.solarsynk_X_inverter_voltage    0
sensor.solarsynk_X_inverter_current    0
sensor.solarsynk_X_inverter_power      0
sensor.solarsynk_X_inverter_voltage    0

Im on HA Core 2024.11.1, Supervisor 2024.11.2, OS 13.2 and Frontend 20241106.2

martinville, 2024-11-11:
Hi Graham,

Yes you are quite right. There is indeed a mistake in the script it sends
all of them to the same entity ending with power voltage and current 2
which explains why you see the 3rd phase only.

Its an easy fix, will correct ot tomprrow afternoon. Will let you know when
its done.

martinville, 2024-11-12:
The issue should now be corrected. thanks for letting me know about the issue.

GrayC88, 2024-11-12:
It is indeed corrected, I'm glad it was a quick fix..

closed 2024-11-12 by GrayC88