martinville/solarsynkv2 #24: Missing entity's

hughlang1978, 2024-12-03:
Import of inverter shows on the log's but no entity's.
Can you please advise

lthorpe18, 2024-12-03:
I'm having the same issue. No errors within the add-on but no entities are appearing withing Home Assistant

Rick-EV, 2024-12-08:
> Import of inverter shows on the log's but no entity's. Can you please advise

In my intro setup had the same issue, solution is to Enable_Verbose_Log in configuration, then going to log and at the last readout you can read all available entities.

vaughanwill, 2024-12-10:
Aame issue here - no entities available. Once resolved pleaae could you also add the link to slipx dataflow card.yaml to your doc repository ? 
[image]

gta80, 2024-12-23:
I have the same problem, I see that all the data is received, but then - 
/run.sh: line 513: [: ==: unary operator expected
Helper entity input_text.solarsynk_XXXXXXXX_inverter_settings has no value. Therefore no inverter setting will be sent for change.

ElectricMolehill, 2025-02-13:
I got the same error when I had Home Assistant hostname or IP address wrongly configured in the configuration settings with an http:// prefix. The field needs just a bare IP address or a DNS name which resolves to one e.g. homeassistant.local

Dpghill, 2025-02-28:
im getting the same issue.

"inverter_settings has no value"

 i see the data in the logs. IP address and port are correct

[image]