martinville/solarsynkv2 #27: funny mappings for my Sunsynk

vdpollmza, 2025-01-03:
Hello,

I have installed your add on, and you have done great work.

however, I have 3 PV strings, and am trying was trying to figure out why if i setup the the MPPT to 3 and use the PV3 inputs, that i get strange duplicate readings on PV1 and PV3 and PV2 and PV4.

on closer inspection, it seems the entities of PV3 are mapped to the PV1 sensors, and PV4 are mapped to the PV2 sensors, but i cannot for the life of me see where to change it in HAS.

do you have any ideas?

regards
Marc

martinville, 2025-01-03:
Hi , Thanks for letting me know this is now corrected in update .21

vdpollmza, 2025-01-03:
You are a legend.

may i ask, was it in run.sh where you made the changes?

vdpollmza, 2025-01-03:
> I Made changes to line 209 to 214
>
> From
> pv3_current=$(jq -r '.data.pvIV[0].ipv' pvindata.json); if [ $pv3_current == "null" ]; then pv3_current="0"; fi;
> pv3_power=$(jq -r '.data.pvIV[0].ppv' pvindata.json); if [ $pv3_power == "null" ]; then pv3_power="0"; fi;
> pv3_voltage=$(jq -r '.data.pvIV[0].vpv' pvindata.json); if [ $pv3_voltage == "null" ]; then pv3_voltage="0"; fi;
> pv4_current=$(jq -r '.data.pvIV[1].ipv' pvindata.json); if [ $pv4_current == "null" ]; then pv4_current="0"; fi;
> pv4_power=$(jq -r '.data.pvIV[1].ppv' pvindata.json); if [ $pv4_power == "null" ]; then pv4_power="0"; fi;
> pv4_voltage=$(jq -r '.data.pvIV[1].vpv' pvindata.json); if [ $pv4_voltage == "null" ]; then pv4_voltage="0"; fi;
>
> changed to
>
> pv3_current=$(jq -r '.data.pvIV[2].ipv' pvindata.json); if [ $pv3_current == "null" ]; then pv3_current="0"; fi;
> pv3_power=$(jq -r '.data.pvIV[2].ppv' pvindata.json); if [ $pv3_power == "null" ]; then pv3_power="0"; fi;
> pv3_voltage=$(jq -r '.data.pvIV[2].vpv' pvindata.json); if [ $pv3_voltage == "null" ]; then pv3_voltage="0"; fi;
> pv4_current=$(jq -r '.data.pvIV[3].ipv' pvindata.json); if [ $pv4_current == "null" ]; then pv4_current="0"; fi;
> pv4_power=$(jq -r '.data.pvIV[3].ppv' pvindata.json); if [ $pv4_power == "null" ]; then pv4_power="0"; fi;
> pv4_voltage=$(jq -r '.data.pvIV[3].vpv' pvindata.json); if [ $pv4_voltage == "null" ]; then pv4_voltage="0"; fi;

thank you for letting me know. was looking at that and was wondering whether it could be that. thank you for confirming - i learned new things today :-)

closed 2025-01-30 by martinville