martinville/solarsynkv2 #30: I was unable to find my Serial Number, but this may help others

shaeney1, 2025-01-18:
Great work!

I installed your Add-on, but wasn't sure where to find my serial number. It wasnt the SN off the Dongle attached to the inverter.

Using the Sunsynk web app, I navigated to my "Plant", Equipment and had to navigate to the Inverter and the value we need is the "Alias", which for me is a 10-digit number.

You can also create a script to hit the following (with the bearer token):

https://api.sunsynk.net/api/v1/plants?page=1&limit=10&name=&status= 

Then  call the plant specific url with your plant id taken from above:

https://api.sunsynk.net/api/v1/plant/<your_current_plant_id>/inverters

This will return you json, which includes both the Alias and the SN and you can see they are the same thing. I assume a future update could be to iterate over the Plant and automatically extract the SN, rather than needing them to be entered?

Hope this helps anyone else.

martinville, 2025-01-24:
Thanks for this I will certainly take a look. ;-)

martinville, 2025-03-21:
I'm closing this issue for now. I’ve looked into it, and while it's certainly possible, the main concern is that it would reduce flexibility by forcing the user to download all inverters into a fixed set of entities.

As a result, I won’t be implementing this. I’d prefer to leave it up to the user to manually select which serial numbers to download by providing a list.

closed 2025-03-21 by martinville