martinville/solarsynkv2 #31: Solarsynk returning alternate null and then  correct readings every 2 minutes

Selseymech, 2025-01-20:
Solarsynk add-on is returning alternate null and then  correct readings every 2 minutes.I have reinstalled installed it and re checked all log in criteria but no difference.Any help would be much appreciated.TIA.

vdpollmza, 2025-01-20:
i have a similiar issue. all my readings are null or zero, but mine takes longer as my refress is every 300 seconds. will try to put it onto 60seconds and see. if I restart the solarsynk add-on, I get the readings again. started yesterday with me.

Selseymech, 2025-01-20:
Hi and thanks for the e mail.My refresh rate at the moment is 60s for both Solarsynk and Sunsynk gateway but I have tried 300s with same result.Only started playing up a few days ago and before was faultless and a very useful app!Sent from my iPad

vividgrd, 2025-01-20:
I'm also seeing the same problem. See issue 32 for request of mitigation approach. 

It would be good to find a solution. My guess is something to do with SunSynk's interface.

Selseymech, 2025-01-20:
I suspected the same but the readings from the Sunsynk app and from Sunsynk.net are always ok?Sent from my iPad

vividgrd, 2025-01-21:
If I get some time over the next few days I might fork this project and modify it so we have a flag to indicate if reads are valid. Setting values to 0 is disruptive for many automations.

I still suspect the issue is with Sunsynk but it has highlighted an issue with the current project. Although it's a fantastic project overall.

Selseymech, 2025-01-21:
Sounds good, thanks for the heads up!

vdpollmza, 2025-01-23:
Seems to have been resolved? I am not seeing this now. So it would seem that there was something on the backend at Sunsynk?

vividgrd, 2025-01-23:
It also seems resoved for me. It's highlighted an issue with this module though. I think having a null/OK data flag would be useful to avoid zero values affecting automations

Selseymech, 2025-01-23:
Good morning
     Yes it does seem solid now thanks to Morne Saunders re configuring his
Solar-Sunsynk integration.(see his GitHub page) Although separate
integrations he believes there could possibly be some interaction between
the two.So yes mine has been solid since his update of Solar-sunsynk so
well done and thanks to him-I hope yours stays solid too.
Best regards
Geoff Mellett

Selseymech, 2025-01-23:
Further to my e mail 1 hr ago the solarsynk integration is back to occasional null readings.I am monitoring the Soloar sunsynk integration which is reading ok and stable??

closed 2025-01-30 by martinville

Selseymech, 2025-01-30:
Thank you Martin all seems solid now!GM

Closed #31 as completed.