martinville/solarsynkv2 #32: Data reads are suddenly returning many null

vividgrd, 2025-01-20:
Starting on 19th Jan 2025, many reads are now returning null, which gets returned as 0 values in the data. 

There seems to be no obvious reason for this. I've tried to restart HA just in case. 

Moreover, it highlights a robustness issue when data is getting returned as 0. This causes various automations to trigger when they shouldn't. e.g. battery SOC suddenly becoming 0 triggers a lot of stuff to shutdown. 

The overall state entity gets set to zero with null reads. It would be more useful if that could also be set to a valid value e.g. 1 on good reads. Then it could be used to validate the other data.

uwalcher, 2025-01-21:
Hi 
I have the same Issue. Seems to start on the 18th of January. Before it was stable.

[image]

Yaheli003, 2025-01-21:
Same problem here.

uwalcher, 2025-01-22:
Problem seems solved. 

[image]

jon0881, 2025-01-24:
the problem is back again for me

martinville, 2025-01-24:
I haven't used this integration for quite some time, but seeing there are now a few people who has this issue I will take a look to see exactly why this is happening and report back.

martinville, 2025-01-24:
For some reason, when requesting an API key from the Sunsynk server, it behaves inconsistently and doesn't always return a valid key. To address this, I implemented a loop that repeatedly queries the server until a valid token is received.

Please test and let me know if it works you can just update the addon via the update button.