martinville/solarsynkv2 #34: Can't get it to work!

lazlosuppel, 2025-02-18:
Hi, I am getting this message-

'Script execution date & time: 18/02/2025 09:54:57
Verbose logging is set to: false
HTTP Connect type: http
Getting bearer token from solar service provider's API.
Invalid token received: Retrying...
Invalid token received: Retrying...'

Can anyone help, I'm new to HA....sorry

HuggiesHA, 2025-02-28:
I am getting the same error

HuggiesHA, 2025-03-13:
Any update on this please?

DSContracts, 2025-03-14:
I’m having the same issue is there any advice please

Hothfix, 2025-03-16:
This happens when you have Multi-factor authentication (MFA) turned on in the SunSynk Connect app. It can't generate a valid token until an OTP has been supplied. Automating this is bound to be tricky. 

**To fix this problem**: 
Click on your account, go to `Personal Settings` > `MFA` and disable it. 

[image]

HuggiesHA, 2025-03-16:
Just checked and my MFA option in my mobile Sunsynk app is disabled and always has been.

Just to confirm as well. The IP is local isn’t it? Don’t need to use DNS or Nuba Casa URL?

Hothfix, 2025-03-16:
Local should be fine, you don't need a Nabu Casa URL. I didn't have to do anything DNS-related when I set mine up on a Pi yesterday. I used `homeassistant.local` for the hostname, set the port, added the token and started it up. I hit the same issue with the invalid token, but disabling MFA fixed it for me. 

I assume you've checked that you're not prefixing the IP with `http://` or `https://`. that you're not using `localhost` or `127.0.0.1` and you've specified the port separately rather than appending it to the hostname? I almost did that :)

martinville, 2025-03-21:
Hi,

When I first started this project, the Sunsynk server used a simple token encryption method. Since then, they have upgraded to a more secure system. The Bash script fetching the token checks its character length to validate it before proceeding. However, this approach is quite crude—mainly due to my limited experience with Bash scripting.

I have now completely migrated this integration to Python, a language I’m much more familiar with. This allows for proper error handling and detailed logs for user and connection errors.

This repository will no longer be maintained. Feel free to check out the new version, which offers better error handling and improved stability:

https://github.com/martinville/solarsynkv3