martinville/solarsynkv2 #36: Invalid token received: Retrying

doart, 2025-04-13:
I had the system running perfectly however had a HA crash and it also fail to install bakups new install I am now get
"Getting bearer token from solar service provider's API.
Invalid token received: Retrying..."

And am at totally lost .
Can someone please help?

martinville, 2025-04-13:
You may want to try the newer version 

https://github.com/martinville/solarsynkv3

doart, 2025-04-13:
I completely deleted Sunsynk2 as suggested with all its data then installed the Sunsynk3.
This was fine except it failed to log in to the Sunsynk account. Login and password are correct these are used to log in my Sunsynk account.
How do I resolve this. Thank-you 

"Sunsynk Login: Account or password error
Error retrieving Bearer Token.
Traceback (most recent call last):
  File "/main.py", line 64, in <module>
    raise ValueError("Failed to retrieve Bearer Token. Check credentials or server status.")
ValueError: Failed to retrieve Bearer Token. Check credentials or server status.
Script execution completed."

martinville, 2025-04-13:
Did you perhaps enable MFA recently ?

doart, 2025-04-14:
Martin I am not too sure what you mean by enable MFA recently.
Is this what you are referring to is no did not enable any thing.
Thank you for your patience with an old man trying and not always without
it.

Best Regards
Art Holder
***@***.***

martinville, 2025-04-14:
Hi I meant MFA "Multifactor Authentication" on your Sunsynk account. When you login to the portal do you have to enter a "Pincode" after providing the password  ?

In the past there have also been some cases where a special character in the password causes issues with t his integration.

doart, 2025-04-14:
Martin I have not been able to resolve this issue as yet will keep
looking and trying.
The Login and the Password is 100% correct, no special characters or
spaces, the only thing I can think of is somehow the Bearer Tokenis
not being retrieved??

Will Keep looking. Again strange prior to the whole crash it worked
great but not with the new install.
Thank you for your assistance
Arthur

doart, 2025-04-25:
Hi Martin
After much frustration I have got it to work without any log errors.
however I cannot get the Solar energy chart populated.
The entities are all visible with values at devices and services/entities i
have also updated the Helper section.
See attached logs what am I now missing.

Many Thanks
Arthur

doart, 2025-04-25:
Logs

-- 
Best Regards
Art Holder
***@***.***